meshtastic/firmware #7735: [Feature Request]: Add indication that node GPS coordinate is not fresh but last known

pedjas, 2025-08-24:
### Platform

Cross-Platform

### Description

When node dos not have GPS module but pulls GPS coordinates from mobile phone, if mobile phone is disconnected, node continues to send last received coordinates.

In some situations this can be confusing, misleading and even cause trouble. For example if one is tracking kid, dog or something else, and phone's battery goes off.

It would be good to have option to set node to behave differently, if it cannot get fresh data from phone. 

One way could be to stop sending coordinates.

Other way could be to send last known coordinates but indicate they are not fresh but last known.

The second option is better because it covers both cases: if node is lost and separated from phone, it will send its location with indication that it is separated from phone, or if phone turns off, node would still send last known coordinate but indicate that it is not updated so it may not be correct, if tracked object is still moving.

fifieldt, 2025-09-21:
Good.

If fixed_position is off, gps is not present, and phone is not connected (for more than position broadcast interval), we probably shouldn't broadcast position.

PositionModule is the likely place that needs updating.

fifieldt, 2025-09-26:
That would require extra bits on every position broadcast, making position less reliable overall :)

thebentern, 2025-09-26:
We do already have timestamp on the position packet. Perhaps there is more behavior we could tailor around that