meshtastic/firmware #7769: [Bug]: Telemetry - no current reading from INA226 module

BurdMahn, 2025-08-27:
### Category

Hardware Compatibility

### Hardware

Station G2

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.11

### Description

I am building a large infrastructure node from a Station G2, a 12V 20AH lead acid battery, and a 20W solar panel and pwm charge controller. As a part of this build I wanted to be able to monitor battery voltage and current draw to keep a close eye on the charging system. I purchased and installed a INA226 sensor (https://www.amazon.com/dp/B0CDWXB8PG) and hooked it up to the G2, with vbus connected to battery positive, and running the main power wire from the battery through the INA226's built in 0.1ohm resistor. The setup seems to work partially, as after enabling the power telemetry module, the sensor shows up, and reports voltage fine, but it will not report current, it just sits there showing 0mA. I can manually probe both sides of the shunt and measure 1.0V, which at 0.1ohms, comes out to 1A of current, so I know everything is hooked up to the sensor correctly. The official documentation lists the INA226 as compatible.

### Relevant log output

```Shell

```

BurdMahn, 2025-08-27:
nevermind, I really shouldve searched the issues better, I see now that this is marked as fixed in the 2.7.4 alpha

Stephan-2025, 2025-09-18:
Yes, worked again with 2.7.4 and greater

closed 2025-09-18 by thebentern