meshtastic/firmware #7770: [Feature Request]: Define a first hop on channel messages

RCGV1, 2025-08-27:
### Platform

Cross-Platform

### Description

Allow users to define the first hop of a channel message using the next hop field. You will be able to define which node will have to rebroadcast first, and then it will flood past that.

Coopersmith-24601, 2025-08-27:
Do you mean first hop or zero hop?  There's a difference.

b8b8, 2025-08-28:
> Do you mean first hop or zero hop? There's a difference.

The part that saves to the packet for next hop routing.

alecperkins, 2025-09-18:
I effect this now by dialing the tx power to 1 dBm on my pocket node when at home, so only my roof node hears it. Otherwise, I sometimes get picked up by a nearby client with less reach that preempts the roof. Fussing with the power works well enough, until I forget to raise it when out and about and wonder why I have terrible reach, or forget to lower it when I return and again have poor reach. Being able to designate a preferred first hop would eliminate the need to go back and forth with the power setting.

FJF130, 2026-02-06:
What is the status of this?

NomDeTom, 2026-03-05:
@RCGV1 what was the conclusion of your testing on this? I know you were fighting issues with failure at the first/last hop for some things.

prokrypt, 2026-08-31:
@RCGV1 🫪