meshtastic/firmware #7771: [Bug]: T-Lora Pager thinks that it's 2005

JThramer, 2025-08-27:
### Category

Hardware Compatibility

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [x] OLED slide UI on any display

### Firmware Version

2.7.6 (main - 26c38ffc8)

### Description

T-Lora Pager thinks that it's 2005-08-27, which leads to very interesting behavior.

This seems somewhat tied to GPS, which performs pretty terribly on the BaseUI on this device. Other devices I own (including ones with no GPS at all) don't seem to have this problem.

Flagging it because it seems to lead to pretty interesting behavior in-app.

### Relevant log output

```Shell

```

JThramer, 2025-08-27:
Said interesting behavior: 

[image] 
("Pager calling" was sent first)

[image]
(my pager is apparently ancient history...)

fifieldt, 2025-08-27:
Looking into this related one: https://github.com/meshtastic/firmware/issues/7750#issuecomment-3225948816

@JThramer most helpful would be logs - from first boot to 30+mins later. Plug in via USB go to https://flasher.meshtastic.org/ and click Open Serial monitor

fifieldt, 2025-08-28:
Thank you very much @JThramer .

This makes it clear the time is coming from the RTC chip, not the GPS as I assumed.

 [0m[34mDEBUG [0m| ??:??:?? 3 [34mRead RTC time from PCF8563 getDateTime as 2005-08-29 06:33:14 (1125297194)

More patches incoming!

PR #7779 (2025-08-29): Can't trust RTCs to tell the time.
Further to https://github.com/meshtastic/firmware/pull/7772 , we discovered that some RTCs have hard-coded start times well in the past.

This patch gives RTCs the same treatment as GPS - if the time is earlier than BUILD_EPOCH, we don't use it.

Fixes #7771
Fixes #7750
commit 10c68362 (2025-08-29): Can't trust RTCs to tell the time. (#7779)

Further to https://github.com/meshtastic/firmware/pull/7772 ,
we discovered that some RTCs have hard-coded start times well in the
past.

This patch gives RTCs the same treatment as GPS - if the time is
earlier than BUILD_EPOCH, we don't use it.

Fixes #7771
Fixes #7750
commit 11db6d4d (2025-08-29): Can't trust RTCs to tell the time. (#7779)

Further to https://github.com/meshtastic/firmware/pull/7772 ,
we discovered that some RTCs have hard-coded start times well in the
past.

This patch gives RTCs the same treatment as GPS - if the time is
earlier than BUILD_EPOCH, we don't use it.

Fixes #7771
Fixes #7750
closed 2025-08-29 by thebentern