meshtastic/firmware #7791: [Bug]: Lilygo T-Beam 1.1 - SX1268 - Critical Fault #3

gandinin, 2025-08-30:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.10 Stable and 2.6.11 Beta

### Description

My LilyGO T-Beam 433 MHz with SX1268 fails to initialize the LoRa radio when flashing Meshtastic firmware (both stable and beta).
At boot I always get No SX1268 radio and Critical error #3.
Hardware:

Device: LilyGO T-Beam (classic, ESP32, not S3)
Radio module: SX1268 (confirmed printed on chip)
Frequency: 433 MHz (label on board)
Battery: 18650 installed
Connected via USB on macOS

Firmware tested:
2.6.10 Stable (from Web Flasher) → fails
2.6.11 Beta (commit 60ec05e) → fails

[image]

### Relevant log output

```Shell
INFO  |  SX126x init result -707
WARN  |  No SX1268 radio
ERROR |  Record critical error 3 at src/main.cpp:1279
```

fifieldt, 2025-08-30:
Can you try changing the power in Lora settings to a lower value?

gandinin, 2025-08-30:
Thanks for rely.
Tried down to zero and up to 12 dbm, nothing change :(

closed 2025-11-16 by github-actions[bot]