meshtastic/firmware #7821: [Bug]: Rotary Encoder is still poor on T-Lora Pager

V8Joel, 2025-09-02:
### Category

Hardware Compatibility

### Hardware

T-Lora v2 1.6

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.8

### Description

Having built the latest develop branch (as of 1/9/2025), I tried Meshtastic on my new T-Lora pager V2. The rotary encoder performance is still poor. It’s still really hard to select menus and scroll around. Probably a 30% success rate. I tried MeshCore on the same device and that works perfectly. Maybe we can see how it’s handled in that stack and gain some inspiration?

### Relevant log output

```Shell

```

mverch67, 2025-09-02:
I guess you scroll too fast so that it skips the states of the state machine?
The wheel can be turned step by step, try this out.

V8Joel, 2025-09-02:
> I guess you scroll too fast so that it skips the states of the state machine?
> The wheel can be turned step by step, each step generates an event to the UI. Try this out.
> Also make sure that you did a factory reset in case you ever flashed any previous firmware from the PR branch.

Thanks for your reply. I have tried this. You have to be incredibly precise about how slowly you turn the wheel, and even then it often misses or overshoots. If you try MeshCore for comparison you'll see how much better it's handled on there. You can turn it as fast as you want and it's pretty much 100% reliable. Vast difference.