meshtastic/firmware #7839: [Bug]: Routing issue - nextHop/next_hop seems to be set by the return traceroute path; it should be set from the forward path

mihvoi, 2025-09-03:
### Category

Other

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

 2.6.11.60ec05e

### Description

nextHop is wrongly set for asymmetric routes.
nextHop should be set from the forward traceroute path, however it seems to be set from the return path.

This makes Meshtastic unresponsive for many users, some are moving to meshcore.

For example, my traceroute from A to B is:
forward: A -> C ->  B
return: B -> A

A can hear B, however B cannot hear A.
A needs C as relay in order to reach B.
Currently, the nextHop seems to be set to A, while it should be set to C - should be taken from the forward path. 

This scenario is very common when B is well positioned on a building, while A is in a remote building where it has many reflections on transmit.

### Relevant log output

```Shell
INFO | 10:15:32 86516 [Router] Received traceroute from=0x5c1cb045, id=0x12f2375a, portnum=70, payloadlen=19

INFO | 10:15:32 86516 [Router] Route traced:

0xda742564 --> 0xd2079685 (6.50dB) --> 0x5c1cb045 (-7.25dB)

(-15.75dB) 0xda742564 <-- 0x5c1cb045

DEBUG | 10:15:32 86516 [Router] Module 'traceroute' considered

DEBUG | 10:15:32 86516 [Router] Module 'neighborinfo' wantsPacket=1

INFO | 10:15:32 86516 [Router] Received neighborinfo from=0x5c1cb045, id=0x12f2375a, portnum=70, payloadlen=31

DEBUG | 10:15:32 86516 [Router] Module 'neighborinfo' considered

DEBUG | 10:15:32 86516 [Router] Module 'routing' wantsPacket=1

INFO | 10:15:32 86516 [Router] Received routing from=0x5c1cb045, id=0x12f2375a, portnum=70, payloadlen=31

DEBUG | 10:15:32 86516 [Router] Routing sniffing (id=0x12f2375a fr=0x5c1cb045 to=0xda742564, WantAck=1, HopLim=3 Ch=0x0 Portnum=70 requestId=448cbc81 rxtime=1756721732 rxSNR=-15.75 rxRSSI=-1

DEBUG | 10:15:32 86516 [Router] Partially randomized packet id 4064789965

DEBUG | 10:15:32 86516 [Router] localSend to channel 0

DEBUG | 10:15:32 86516 [Router] Add packet record fr=0xda742564, id=0xf247c5cd

DEBUG | 10:15:32 86516 [Router] Setting next hop for packet with dest 5c1cb045 to 0

DEBUG | 10:15:32 86516 [Router] Expand short PSK #1

DEBUG | 10:15:32 86516 [Router] Use AES128 key!

DEBUG | 10:15:32 86516 [Router] enqueue for send (id=0xf247c5cd fr=0xda742564 to=0x5c1cb045, WantAck=0, HopLim=0 Ch=0x8 encrypted len=29 rxtime=1756721732 relay=0x64 priority=120)

DEBUG | 10:15:32 86516 [Router] txGood=271,txRelay=0,rxGood=6805,rxBad=719

DEBUG | 10:15:32 86516 [Router] Received a ACK for 0x448cbc81, stopping retransmissions

INFO | 10:15:32 86516 [Router] Update next hop of 0x5c1cb045 to 0x45 based on ACK/reply
```

GUVWAF, 2025-09-03:
Thanks for the report. This is indeed wrong and seems to be caused by this logic: 
https://github.com/meshtastic/firmware/blob/789c1ab59d4aa91a6f24b9e6f07ef0a78e447e44/src/mesh/NextHopRouter.cpp#L77

This is meant for the case when we were the last relayer in a route and you get the ACK/reply directly from the destination. However, when we sent out the traceroute, we set ourselves already as one of the relayers and thus this would be true, even though it actually went via 0xd2079685.

I will need to think deep about this, but I believe this can be fixed by checking if we were the *only* relayer.

mihvoi, 2025-09-03:
Without fully understanding how 2.6 routing was designed, I guess we can extract more than one nextHop from a traceroute:
- If a B sees traceroute ack from E to A with forward path A->B->C->D->E, 
            then B can set C as nextHop for D and E (not only for E) 
- Similarly, A can set B as nextHop for C,D and E.

mihvoi, 2025-09-06:
Thanks. Looking forward for merge&release.

PR #7859 (2025-09-06): Upon receiving ACK/reply directly, only update next-hop if we’re the *sole* relayer
Fixes #7839. 

Simulated the same scenario with the [interactive simulator](https://github.com/meshtastic/Meshtasticator/blob/master/INTERACTIVE_SIM.md) and it works correctly now. 
Previously it indeed set the next-hop while it shouldn’t: 

```
INFO  | 19:44:23 12 [Router] Received traceroute from=0x12, id=0x21ceb01c, portnum=70, payloadlen=28
INFO  | 19:44:23 12 [Router] Route traced:
0x10 --> 0x11 (-7.00dB) --> 0x12 (-7.25dB)
(-12.00dB) 0x10 <-- 0x12
INFO  | 19:44:23 47 [Router] Update next hop of 0x12 to 0x [... cut]
closed 2025-09-08 by fifieldt