meshtastic/firmware #7863: [Feature Request]: CLIENT_BASE mode: ROUTER for favorites, CLIENT otherwise (for attic/roof nodes!)

compumike, 2025-09-04:
### Platform

Cross-Platform

### Description

Like many Meshtastic fans, I currently have a roof/attic node (`CLIENT` mode), plus one or more mobile nodes (`CLIENT_MUTE` or `CLIENT`) which are indoors and/or much less well positioned. **The current modes and routing algorithm do not let me take advantage of my relatively strong roof/attic node.**

Due to managed flooding SNR prioritization, my mobile node’s outgoing packets are usually picked up and rebroadcasted by a relatively weak (probably indoor) neighbor node elsewhere in my town, not by my much better-positioned base station.

-----

### Proposal: `CLIENT_BASE` mode (for attic/roof nodes)

I propose a new `CLIENT_BASE` mode, to be suggested for possible use on attic/roof nodes, with the following behavior:

1. Acts as `ROUTER` (early, guaranteed rebroadcast) for any packets where either `p->from` OR `p->to` are an `is_favorite` node.
2. Acts as `CLIENT` otherwise.

(Seriously, that's it.)

-----

### Super Easy Implementation

Adding this `CLIENT_BASE` mode requires:

- no extra firmware state
- no changes to packets
- very few lines of code

In fact, it might just be one extra `else if` condition in `RadioInterface::getTxDelayMsecWeighted`: :smile:

https://github.com/meshtastic/firmware/blob/f31fd34ce049d4720a2adc69101fb9fe2e8d1b62/src/mesh/RadioInterface.cpp#L325

-----

### Actually Uses Attic/Roof Nodes

With attic/roof nodes set to `CLIENT_BASE`, those stronger nodes will actually get used by the routing! This may encourage people to actually deploy more of these stronger nodes.

-----

### Improves Packet Delivery Reliability

Currently, packets from my relatively weak indoor node are most likely getting rebroadcast by some other relatively weak indoor node.

With `CLIENT_BASE`, messages sent from a weaker/indoor/mobile device will actually take advantage of their nearby attic/roof base station, which currently does not get utilized reliably or at all. This may significantly extend the range at which messages can be reliably exchanged in a single hop.

-----

### Falls Back Perfectly

When not near a CLIENT_BASE, the mobile CLIENT/CLIENT_MUTE just works as normal.

-----

### Congestion Neutral (maybe even negative!)

The change only applies to packets from/to a handful of personally favorited nodes. In fact, because the **stronger** `CLIENT_BASE` does an early rebroadcast, it can suppress (potentially multiple) later `CLIENT` rebroadcasts from weaker radios, since the strong `CLIENT_BASE` may be able to transmit much further.

In my current situation: there's currently one packet from my mobile node to some relatively weak/indoor neighbor node `CLIENT` which is not under my control, plus that node's rebroadcast (which doesn't go very far). Two packets total.

With `CLIENT_BASE`: there's one packet from my mobile node, which is then quickly rebroadcast (as `ROUTER`) by my stronger attic/roof node (set to `CLIENT_BASE`, with my mobile node as a favorite). This stronger rebroadcast will be received by my weak neighbor node and suppress its rebroadcast. So again: two packets total, but one is actually from a strong/well-placed station.

-----

I’m open to writing a PR here. Looking for feedback as to whether this would be something the core team would consider, and whether other Meshtastic users might find it valuable. Thank you!

fifieldt, 2025-09-04:
Thanks for writing in :)

Have you had a look at ROUTER_LATE ?

compumike, 2025-09-04:
@fifieldt Thanks for taking a look! Yes, I have had a look at `ROUTER_LATE`. It has potential. But also seems to have a lot more congestion impact:

1. `ROUTER_LATE` has a much much higher congestion impact due to always-forward-all-packets behavior. (Which is why `ROUTER` is generally discouraged -- same should apply to `ROUTER_LATE`, I think.)
2. `ROUTER_LATE`, by virtue of being late, would not suppress the relatively weak neighbor node in my example from doing its earlier `CLIENT` broadcast, leading to yet more congestion.

Would the current design encourage people to generally set their roof/attic nodes to `ROUTER_LATE`?

Given my reading, I didn't think so. But this new `CLIENT_BASE` mode could be safely encouraged widely.

Kealper, 2025-09-07:
I read the title and was initially thinking "oh no, not another role..." but as soon as I read what you were proposing, I really like this idea.

My area's mesh is growing rapidly and I've already noticed some newer people wanting to pick ROUTER for their rooftop nodes just so it gets sending priority over their vehicle nodes that are in CLIENT and I've been pushing them to at least set the rooftop node to ROUTER_LATE but I don't like that as a permanent fix for these people either because although it fixes the routing issues, each one of those adds a little bit of unnecessary load on our local mesh.

This role would solve that completely and if this gets implemented, I'll gladly start coordinating people in my local mesh to switch some nodes over to the alpha release that includes support for it and give the role a real-world test.

ForeheadFox, 2025-09-08:
My use case: home node (set as client) sometimes hears a packet but doesn’t forward it to my mobile node I’m connected to, so the message gets lost. This also affects several people in my mesh, we end up “losing” messages unless we set nodes to Router in poor locations, which isn’t ideal.

erayd, 2025-09-09:
I'm definitely a fan of this suggested role. It both serves a useful purpose, and helps discourage people from running infrastructure roles in inappropriate places. 

One caveat to note: this new role does not handle incoming broadcast messages that are not from a favourite nodes. So some messages should be expected to be lost due to this.

compumike, 2025-09-09:
> One caveat to note: this new role does not handle incoming broadcast messages that are not from a favourite nodes. So some messages should be expected to be lost due to this.

@erayd Thanks for pointing this out. You're correct that the current proposal helps more with **outgoing** broadcast messages, or DMs bidirectionally, but doesn't do anything special at the moment for **incoming** broadcast messages.

Due to the SNR-based CLIENT rebroadcasting rules, currently:

1. If the `CLIENT_MESH` base station and `CLIENT` indoor node both hear the incoming broadcast packet, the weaker `CLIENT` is likely to have worse SNR and therefore will win the rebroadcasting race. (This is less-than-ideal.)
2. If only the `CLIENT_MESH` hears the incoming broadcast packet, it will rebroadcast. (Great.)
3. If only the `CLIENT` hears the incoming broadcast packet, it will rebroadcast with the original `p->from`. And again, the `CLIENT_MESH` will be unlikely to rebroadcast if a second `CLIENT` hears the first `CLIENT`'s rebroadcast.  (This is less-than-ideal.)

I suspect 1 and 2 are more common than 3.

We could theoretically modify the proposal to have `CLIENT_BASE` also act as `ROUTER` for any incoming broadcast packets, however this is likely to have a **much** higher impact on congestion, which I'd guess makes this not viable.

Open to suggestions.

erayd, 2025-09-09:
> Open to suggestions.

What if it used a *VERY* low power to repeat everything incoming, using the same timing as ROUTER_LATE? So things in the house could hear it, but not too much else. And then full power to rebroadcast anything from a favourite node, with the fast timing?

Kealper, 2025-09-09:
> What if it used a VERY low power to repeat everything incoming, using the same timing as ROUTER_LATE?
I'm against this idea, I think the original implementation of this role is the correct implementation.

With the original implementation, it allows people who go through the work to build a nice base station with much better coverage to utilize that investment without having to set situationally-inappropriate roles to guide their setup to route in the way they want without causing issues for others.

If the base station and, for example, a vehicle repeater (set to CLIENT), both hear another node consistently but it's determined that the base station being the third-party node's hop is actually beneficial, that third-party node could be added to the favorites list on the base station in the original implementation. If there's multiple third-party nodes that actually benefit from the base station in question doing the rebroadcasting, then this new role would not be the best fit, and ROUTER_LATE could actually be the better option. I feel that the most common problem that needs fixing with this new role is where the mesh in the area is already set up enough that if the base station always rebroadcasts the base station owner's packets, then the owner gets to consistently have a strong connection into their local mesh.

b8b8, 2025-09-09:
> I'm against this idea, I think the original implementation of this role is the correct implementation.
>
> With the original implementation, it allows people who go through the work to build a nice [...]

Good points, i agree. There is also the fact that in general nodes are better at receiving than transmitting so the hardest part is getting out, not getting packets into the house. The few missed incoming channel packets are less of an issue when in/out DMs and outgoing channel packets are guaranteed without the risk of yet another forced rebroadcast role.

eureekasigns, 2025-09-13:
Fascinating idea. 
Some questions. 

First, does this apply to trace route as well as direct messages? I may be too near another node (can occasionally directly receive) but am seemingly preferring a neighbor client node vs mine when tracing route. Said distant node is also a strange challenge to reliably message, though, and I'm not sure why. Hoping this role can help. It's not that far away (about half a mile)

Second, does the base need distant nodes as favorites, or should it only be your own portables?

Third, is there a limit to number of favorites?

tedwardd, 2025-09-15:
Food for thought: Should `CLIENT_BASE` nodes avoid decrementing the hop count for messages originating from favored nodes? I mention this because the idea of `CLIENT_BASE`, at least in my mind, is to act as a transparent gateway for predefined clients within a given area. On the TX side, this would be relatively straight forward. Simply forward messages from predefined clients with the same hop count as it was received with. The part I can't yet wrap my head around is the RX side due to the broadcast nature of the relayed messages. Perhaps allowing the relaying of messages at the end of their hop limit to be transmitted one last time when in `CLIENT_BASE` mode would work, though I don't know if that's a good idea or not.

b8b8, 2025-09-15:
> Food for thought: Should `CLIENT_BASE` nodes avoid decrementing the hop count for messages originating from favored nodes? I mention this because the idea of `CLIENT_BASE`, at least in my mind, is [...]

Probably too dangerous and early for this new role to be messing with hop counts (for now).

Syridian, 2025-11-13:
https://github.com/meshtastic/firmware/discussions/6736

31337-4554551n, 2026-01-09:
Sorry, I can't quite understand what the status of this is.
Has it been closed as not happening? Or is this closed because it client_base will be router for favourites on the next firmware?
What's the consensus from the devs here?