meshtastic/firmware #7881: [Feature Request]: OTA over LoRa

realtag-github, 2025-09-06:
### Platform

Cross-Platform

### Description

I had an idea about how to do LoRa OTA. I don't know if it is feasible though.

1. Cryptographically sign firmware binaries. Allow to configure what cryptographic key-pair to accept.
2. Split partition microcontroller into two. Similar to how wifi OTA for esp32 works.
3. Store slow LoRa OTA broadcasts into second partition to start assembling firmware out of chunks. Similar to how torrents or bitcoin blockchain sync work. (Too computationally heavy for microcontroller?) 
4. Have a mechanism that can request chunks from specific node(s). Store OTA firmware on the second partition on the sender node. OTA would be cryptographically signed by Meshtastic, but allows custom key to sign. 
5. Allow nodes to not re-broadcast this type of traffic or limit it only favorite nodes. 
6. Have a rate limit of this type of traffic on every node. If network is too busy then de-prioritize it. 

Maybe I missed something, but I would like to start this discussion. Maybe my idea is too difficult or too computationally heavy for the microcontroller. But to me it looks like it could work.

garthvh, 2025-09-06:
Doing firmware updates over the mesh is just not feasible without destroying the mesh for everyone.

closed 2025-09-06 by garthvh

b8b8, 2025-09-06:
https://github.com/meshtastic/firmware/issues/5870

Your best bet is maybe Wifi HaLow paired with a raspberry pi.