meshtastic/firmware #7884: [Feature Request]: Add bits for important messages (to enable wake-up)

FFAMax, 2025-09-07:
### Platform

Cross-Platform

### Description

Hello, Team.

While testing the mobile setup I found a few very useful features to have.
By default, the module is awake all the time regardless of the charge level and role. BT is also enabled all the time.

Now, let's imagine the module installed on the mountain. During winter the solar panel gets less power, so we need to handle that.
The idea is: implement autonomous mode. In this mode, the battery is calibrated to know the real bottom level. If discharged below ~30%, then put the MCU in sleep mode and put the LoRa module in RX-only mode. Find a way to track magic packets to wake up the MCU (like specific bits, etc.) on the LoRa module by itself (similar to how Ethernet Wake-on-LAN works), or wake up the MCU each time for processing only received packets. In this mode we can skip handling databases and only relay specific data, like emergency signals, etc.

If charging (daylight time + direct sun is available), then the mode can be switched off and the system can operate normally.
Based on battery capacity and setup consumption, the threshold can be adjusted from the default 30% to something else, or calculated automatically after battery calibration.

We can also adjust transmit power if direct peers can be reached with less power.

The goal is to keep the module alive 24/7 regardless of battery level and solar activity: programmatically handle dynamic environments to save the setup from dying.

RFC

b8b8, 2025-09-07:
https://github.com/meshtastic/firmware/issues/6348

https://github.com/meshtastic/firmware/discussions/4616

roberthadow, 2025-10-03:
https://github.com/orgs/meshtastic/discussions/283 Basically the same request.

FFAMax, 2025-10-04:
This request differs from the previous one, which focused solely on entering sleep mode and resuming operations once solar activity is restored.

In contrast, this request is about continuing operation while reducing resource consumption, prioritizing only critical functionality. A separate request (https://github.com/meshtastic/firmware/issues/8212) has also been raised to implement Quality of Service (QoS) handling. This would allow the system to ignore non-critical messages, and QoS can be leveraged in this context as well.

The core idea here is to enable the radio module to recognize incoming traffic before waking up the main IC, allowing it to conserve energy while still maintaining essential communications. 
@thebentern could you please reopen this request. Thanks!

roberthadow, 2025-11-30:
It looks like there is no interest in this feature by the development team.

 [283](https://github.com/orgs/meshtastic/discussions/283)