meshtastic/firmware #7970: [Bug]: Location displayed as DEC when configured as MGRS

RichieFrame, 2025-09-13:
### Category

Other

### Hardware

T-Echo

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.8.a0c0388

### Description

on baseui, when the display location format is configured as MGRS, the display shows LAT/LONG in decimal format instead

### Relevant log output

```Shell

```

Xaositek, 2025-09-13:
Let me see what I can do here!

Xaositek, 2025-09-13:
I don't want to claim to be a GPS expert, but I do believe I've got it working with all the various formats. Feel free to pull the branch from the linked fix and give it a shot.

Xaositek, 2025-09-13:
I had an issue with Easting and Northing values running over the Compass so I had to break them up. I don't know the answer to this but is there a better way to handle that?

Xaositek, 2025-09-13:
Without trying to dox myself, here's an update which improves how it displays with some labels. Google was my friend on this one.
[image]
[image]

RichieFrame, 2025-09-13:
I can see why your display has issues with the full format...

Adding additional letters breaks the format, but splitting it away from the grid square would be OK, for example:

Grid: 15T VL
99945 99912

RichieFrame, 2025-09-13:
Same deal with UTM and OSGR of course

closed 2025-09-13 by HarukiToreda