meshtastic/firmware #8012: [Feature Request]: Heltec MeshPocket battery curve work

shortwavesurfer2009, 2025-09-17:
### Platform

NRF52

### Description

I just purchased a Heltec MeshPocket and gave it a plum full charge (4.47V) until the green light remains solid. However i have had it unplugged for 2 hours now and it is still reporting as being connected to AC power and not showing a battery %

fifieldt, 2025-09-18:
Suggested reading: https://github.com/meshtastic/firmware/pull/7878

shortwavesurfer2009, 2025-09-19:
> Suggested reading: [#7878](https://github.com/meshtastic/firmware/pull/7878)

Hey, thanks. Since that's the default, if I use 4.916 manually or by using the default, I get a voltage of 4.41 right now and as high as 4.47 on the Heltec MeshPocket. To get it anywhere close to 4.2 i have to use 4.61ish