meshtastic/firmware #8014: [Bug]: Broadcast intervals not respected

groveld, 2025-09-17:
### Category

Other

### Hardware

Rak4631

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.7.5ae4ff9

### Description

The broadcast intervals are not respected. I've used the cli commands below;

--set device.node_info_broadcast_secs 10800
--set telemetry.device_update_interval 10800
--set telemetry.environment_update_interval 10800
--set telemetry.power_update_interval 10800
--set position.position_broadcast_secs 10800
--set position.gps_update_interval 3600

And the config gets updated. I i check meshtastic --info or if i export the config to a yaml file everything looks correct but is it not doing what is set.

Below is a log of the sent packages:

| Timestamp           | From | To        | Type      | Channel  |
|---------------------|------|-----------|-----------|----------|
| 2025-09-17 15:45:20 | GAU1 | Broadcast | Telemetry | LongFast |
| 2025-09-17 15:45:14 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 15:09:13 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 14:33:12 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 13:57:19 | GAU1 | Broadcast | Telemetry | LongFast |
| 2025-09-17 13:57:12 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 13:56:45 | GAU1 | Broadcast | Node Info | LongFast |
| 2025-09-17 13:21:11 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 12:45:10 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 12:09:19 | GAU1 | Broadcast | Telemetry | LongFast |
| 2025-09-17 12:09:08 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 11:33:07 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 10:57:06 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 10:56:45 | GAU1 | Broadcast | Node Info | LongFast |
| 2025-09-17 10:21:18 | GAU1 | Broadcast | Telemetry | LongFast |
| 2025-09-17 10:21:06 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 09:39:05 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 08:57:05 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 08:15:17 | GAU1 | Broadcast | Telemetry | LongFast |
| 2025-09-17 08:15:04 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 07:56:45 | GAU1 | Broadcast | Node Info | LongFast |
| 2025-09-17 07:39:04 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 07:03:03 | GAU1 | Broadcast | Position  | LongFast |
| 2025-09-17 06:27:17 | GAU1 | Broadcast | Telemetry | LongFast |
| 2025-09-17 06:27:03 | GAU1 | Broadcast | Position  | LongFast |

### Relevant log output

```Shell

```

GUVWAF, 2025-09-17:
I agree it's not so intuitive, but there is dynamic scaling applied based on the number of online nodes, see: https://meshtastic.org/docs/overview/mesh-algo/#regular-broadcast-intervals

groveld, 2025-09-17:
Also, does this only scale telemetry and position packages?
Node info seems unaffected.

b8b8, 2025-09-17:
> Also, does this only scale telemetry and position packages? Node info seems unaffected.

Telemetry is scaled. Smart position is not scaled.

GUVWAF, 2025-09-17:
> It scales up the intervals up till 40 active nodes?

Indeed, it starts at 60% of the configured value and from 31 online nodes on, it will be 100%. 

>  Does this exclude the router role?

Indeed, the `ROUTER` role does not do any dynamic scaling, as it higher intervals by default already.

> When does it start to scale down the intervals?

Above 40, it scales according to the formula in the documentation, where the throttling factor varies per LoRa modem preset.

> Also, does this only scale telemetry and position packages?

Indeed, NodeInfo is unaffected.

We need to properly document this.