meshtastic/firmware #8039: [Bug]: Nodes stay `unknown`

ViezeVingertjes, 2025-09-19:
### Category

Hardware Compatibility, Other

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

v2.7.9.70724be

### Description

All nodes appear as unknown in the UI. I have not yet deeply investigated, but the issue consistently affects multiple node types.

Nodes running v2.7.9.70724be display correctly to eachother, while nodes on other versions remain stuck in the unknown state.

### Relevant log output

```Shell

```

b8b8, 2025-09-19:
What UI?
What devices have you tried?
Do you have any screenshots?
Do you have any logs?

ViezeVingertjes, 2025-09-19:
> What UI? What devices have you tried? Do you have any screenshots? Do you have any logs?

- Station G2
- T-Echo
- T-Beam
- T1000e

All the same result, not specific to any UI, User/Node info is never processed it seems, but will have to monitor the logs a bit longer for that.

ViezeVingertjes, 2025-09-19:
In v2.7.9.70724be, it appears that `NodeInfo` packets are not being logged at the default debug level. Across 50+ nodes in my environment, there haven't been any updates to green lock status, metadata, or node names. The only devices showing expected behavior are my own, which run the same version—when a `NodeInfo` is sent, it is processed and appears in the logs as expected.

Confirmed by going back to version 2.7.7.5ae4ff9 Alpha that everything works properly again.

ViezeVingertjes, 2025-09-19:
> I can confirm it works with an old deleted 2.7.7 node.
> I sent "hello" from my old 2.7.7 node and my 2.7.9 node got the message then initiated the nodeinfo exchange unprompted.
> I DID NOT use "exchange user info" in the app.

And you did see your 2.7.7 node name, key etc on the 2.7.9 node?
Because that did not work for me, on any of the devices... so weird...
Will try again tomorrow, but only downgrading seemed to fix it for me.

thebentern, 2025-09-20:
Can you post some of your startup logs that show the lora params like frequency slot?

ViezeVingertjes, 2025-09-20:
> [@ViezeVingertjes](https://github.com/ViezeVingertjes) Were you using the release as-is or with your PR [#7753](https://github.com/meshtastic/firmware/pull/7753) cherry-picked? The reason I ask is because I found a bug ([#8057](https://github.com/meshtastic/firmware/pull/8057)) in that part of the code due to merging and it would explain exactly what you describe. Namely, it would set the channel name to “Custom”, generate a wrong hash and hence cannot decrypt. Communication between 2.7.9 versions would still work, but older firmware not.

It was the plain version from the web flasher, so that isnt included, as far as i know(?)

GUVWAF, 2025-09-20:
Okay, no it's not there yet, so then I'm still clueless.

moverhorst-pitcrew, 2025-09-25:
Good to hear that; i also flashed 2.7.10 yesterday, havent had time to check yet, but will report back if it solved it.

fifieldt, 2025-10-08:
Closing, assumed solved by new firmware :)