meshtastic/firmware #8042: [Bug]: Unable to set security.is_managed to True as part of a serial transaction

ndoo, 2025-09-19:
### Category

Serial

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.7

### Description

Trying to configure a node from .yaml via python (`meshtastic --configure my_config.yaml`) fails if the configuration contains `security:` -> `isManaged: true`.

It looks like as part of committing the transaction at the end, causes serial to drop, which causes the configuration transaction to not commit.

### Relevant log output

```Shell
meshtastic --configure .\MY_919_lucifernet_BDKL.yaml
Connected to radio
INFO file:node.py beginSettingsTransaction line:619 Telling open a transaction to edit settings
Setting device owner to [redacted]
Setting device owner short to [redacted]
Setting channel url to [redacted]
Fixing altitude at [redacted] meters
Fixing latitude at [redacted] degrees
Fixing longitude at [redacted] degrees
Setting device position
Set bluetooth.enabled to False
Set device.led_heartbeat_disabled to True
Set device.node_info_broadcast_secs to 0
Set device.rebroadcast_mode to ALL
Set device.role to ROUTER_LATE
Set device.tzdef to <+08>-8
Set lora.channel_num to 0
Set lora.config_ok_to_mqtt to True
Set lora.hop_limit to 0
Set lora.region to MY_919
Set lora.sx126x_rx_boosted_gain to True
Set lora.tx_enabled to True
Set lora.tx_power to 0
Set lora.use_preset to True
Set position.fixed_position to True
Set position.gps_mode to NOT_PRESENT
Set position.position_broadcast_secs to 0
Set position.position_broadcast_smart_enabled to False
Set position.position_flags to 3
Set power.adc_multiplier_override to 3.325
Set security.admin_key to ['base64:[redacted]=', 'base64:[redacted]=']
Set security.is_managed to True
Set security.private_key to base64:[redacted]=
Set security.public_key to base64:[redacted]=
Set neighbor_info.enabled to True
Set neighbor_info.transmit_over_lora to True
Set neighbor_info.update_interval to 0
Set telemetry.device_update_interval to 0
Set telemetry.environment_measurement_enabled to True
INFO file:node.py commitSettingsTransaction line:633 Telling node to commit open transaction for editing settings
Writing modified configuration to device
WARNING file:stream_interface.py __reader line:207 Meshtastic serial port disconnected, disconnecting... ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
```

MarcM987, 2026-10-25:
Hey, I'm new to the repo and saw that this was flagged as a good first issue and would love to help! However, I proactively tried to reproduce the issue and couldn't seem to replicate the problem. If there was anymore insight on reproduction and you're alright with me working on this, I would love to contribute. 

Environment:
Device: Pico 2W with Waveshare SX1262
Firmware: firmware-pico2w-2.7.7.5ae4ff9.uf2
Meshtastic Version: 2.7.11
OS: MacOS 15.7.7 
(Also similarly test checked on Windows 11, and firmware-pico2w-2.8.1.8e6a88d.uf2)

What I tried:
```bash
% meshtastic --get security.is_managed
Connected to radio
security.is_managed: False
Completed getting preferences
% meshtastic --configure test-managed.yaml 
Connected to radio
Set security.is_managed to True
Set security.admin_key to ['base64:[redacted]=']
INFO file:node.py beginSettingsTransaction line:710 Telling open a transaction to edit settings
INFO file:node.py commitSettingsTransaction line:724 Telling node to commit open transaction for editing settings
Writing modified configuration to device
% meshtastic --get security.is_managed
Connected to radio
security.is_managed: True
Completed getting preferences
```

**'test_managed.py'**
```yaml
config:
    security:
        is_managed: true
        admin_key:
            - "base64:[redacted]="
```

ndoo, 2026-10-26:
I suspect this has already been fixed along the way, let me just close it as I have also not had general instability of `--configure` for a while now.

closed 2026-10-26 by ndoo