meshtastic/firmware #8068: [Bug]: T-Lora Pager

DrBrad, 2025-09-21:
### Category

Other

### Hardware

Other

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.9.70724be Alpha

### Description

After flashing the newest firmware - and version: 2.7.6.834c3c5 Alpha. It prompts me with:
Ok
Got it!
I click ether one and I get requested to select a band - I choose US, or any other. The device then states reboot and its the same menu - I cant get anywhere. I get the same message every time.

In the background I see "Critical Fault #13". I flashed this twice with the same issue.

### Relevant log output

```Shell

```

mverch67, 2025-09-22:
Did you choose the option "Full erase and install"?

DrBrad, 2025-09-28:
- turns out it worked after about an hour later - I did have to reflash after flashing a different OS so I dont really know what the bug wwas.

closed 2025-09-28 by mverch67