meshtastic/firmware #8070: Fix Heltec v4 Power Settings

fifieldt, 2025-09-21:
The Heltec v4 has a 17dB attenuator and a GC1109 power amp. It's not a setup we've encountered before. Our TX_GAIN_LORA setting assumes a linear increase in power on top of SX126X_MAX_POWER, while the power amp behaviour in this case is non-linear, and affected by an attenuator in between.

In #7845 we set SX126X_MAX_POWER to 11 as a compromise to get the code in faster while Heltec does testing to look at the output. We must revisit this to ensure users can obtain the maximum legal power for their devices :)

ndoo, 2025-09-22:
I think setting `TX_GAIN_LORA 11` together with `SX126X_MAX_POWER 11` would then be the compromise to start with.

We (who?) need to start thinking about having TX_GAIN_LORA as a struct array or such to map non-linear gain, hm.

Quency-D, 2025-09-25:
This is the power comparison table of the mass production version.
[image]

fifieldt, 2025-09-25:
Ok, sorry @Quency-D - this is just comparing input to the Power Amp and output of the Power Amp, right?

Or, does it include the attenuator?

Quency-D, 2025-09-26:
> Perfect, thank you. It looks like the patch at [#8107](https://github.com/meshtastic/firmware/pull/8107) will give exactly what we need :)

Awesome, you are so efficient.

closed 2025-09-29 by fifieldt

ndoo, 2026-02-05:
> So - if we set a transmit power level of 15 in the Lora settings on a
> Heltec V4, does that we can expect 25.25 dBm (for the 915mhz version
> according to the chart above), or will we get 15dBm?

When you set a transmit power level of 15dBm you will get 15dBm