meshtastic/firmware #8081: [Bug]: Heltec v3: while pairing the BLE Pin is just shown for <1 Second, if ever

Elektron79, 2025-09-22:
### Category

BLE

### Hardware

Heltec V3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [x] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.9 beta

### Description

Hi,

when i initiate the Pairing of a smartphone or PC, the Pin is only visible for <1 Seconds on the display.
Somtimes nothing is shown

### Relevant log output

```Shell

```

thebentern, 2025-09-22:
I have not been able to reproduce this. Do you have an aggressive screen timeout or router role configured?

Elektron79, 2025-09-22:
No, the Heltec is in Client Role and the display remains on.
It just switches back after <1 second to the „normal view“.

Xaositek, 2025-09-25:
I have attempted and unable to reproduce. Can you retest with 2.7.10 to determine if you can provide reproduction steps? Perhaps a video also may be useful.

closed 2025-12-12 by github-actions[bot]