meshtastic/firmware #8084: [Bug]: T-Echo Lite

J0kerFace, 2025-09-22:
### Category

Other

### Hardware

T-Echo

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.7.9.70724be

### Description

LILYGO T-Echo Lite with installed L76K GPS Module doesn´t seems to work. It shows "No Sats" all the time and does not get a position. 

GPS Mode at configuration ist enabled.

### Relevant log output

```Shell

```

Szetya, 2025-09-22:
Sorry, a configuration from an earlier test version was released.

Szetya, 2025-09-22:
Modification tested and working. :)
```
DEBUG | ??:??:?? 5 [GPS] Probe for GPS at 9600
DEBUG | ??:??:?? 6 [GPS] Trying $PDTINFO (Unicore Family)...
DEBUG | ??:??:?? 6 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...
DEBUG | ??:??:?? 7 [GPS] Trying $PAIR021*39 (Airoha Family)...
DEBUG | ??:??:?? 8 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 8 [GPS] Trying $PCAS06,0*1B (L76K)...
INFO  | ??:??:?? 8 [GPS] L76K detected
DEBUG | ??:??:?? 9 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | ??:??:?? 9 [GPS] No GPS lock
DEBUG | ??:??:?? 9 [GPS] onGPSChanged() pos@0 time=0 lat=0 lon=0 alt=0
INFO  | ??:??:?? 9 [GPS] updatePosition LOCAL pos@0 time=0 lat=0 lon=0 alt=0
```

J0kerFace, 2025-09-24:
Can i temporary fix it by editing der rx, tx and en pin in the app config or any other way?

closed 2025-10-08 by fifieldt