meshtastic/firmware #8106: [Feature Request]: Create a separate task for sensor monitoring?

Quency-D, 2025-09-25:
### Platform

ESP32

### Description

I'm trying to add an SGP41 sensor to my code, but I'm running into a problem: the SGP41 needs to read data once per second for two to three minutes before calculating the IAQ value. This seems difficult to achieve with the current sensor processing architecture. Does anyone have any suggestions?

[image]

oscgonfer, 2025-09-25:
Hi,

Thanks for the ping. For sensors that require a duty cycle, I think the best is to use the `runOnce()` function in the sensor implementation for the AQ module. I applied this on the [SEN5X](https://github.com/fablabbcn/smartcitizen-meshtastic/tree/feat/add-sen5x) sensor (which requires 15s stabilisation). I'd rebase onto the [AQ refurbish PR](https://github.com/fablabbcn/smartcitizen-meshtastic/tree/enhancement/air-quality-module).

I wouldn't think there's a need for a separate thread as the AQ module should be able to handle this. Since you can heat up the sensor and stabilize it for each reading, I'd do, in the sensor `runOnce` function:

1. Trigger the sensor heat up and wait for 200ms.
2. Request measurement data.
3. Return for 1 second to allow for other things and come back to 2) if needed (_I'd review if this is really needed_).
5. Turn off sensor heater.
6. Pass sensor data to the gas index algorithm.
7. Return whatever interval for measuring the sensor. 

For 3), where have you seen this info? @Quency-D

Quency-D, 2025-09-25:
> Hi,
> 
> Thanks for the ping. For sensors that require a duty cycle, I think the best is to use the `runOnce()` function in the sensor implementation for the AQ module. I applied this on the [SEN5X](https://github.com/fablabbcn/smartcitizen-meshtastic/tree/feat/add-sen5x) sensor (which requires 15s stabilisation). I'd rebase onto the [AQ refurbish PR](https://github.com/fablabbcn/smartcitizen-meshtastic/tree/enhancement/air-quality-module).
> 
> I wouldn't think there's a need for a separate thread as the AQ module should be able to handle this. Since you can heat up the sensor and stabilize it for each reading, I'd do, in the sensor `runOnce` function:
> 
> 1. Trigger the sensor heat up and wait for 200ms.
> 2. Request measurement data.
> 3. Return for X seconds (see below) to allow for other things and come back to 2) if needed (_I'd test this_).
> 4. Turn off sensor heater.
> 5. Pass sensor data to the gas index algorithm.
> 6. Return whatever interval for measuring the sensor.
> 
> This would be after the initial break-in period (which would be special case). EDIT: For 3), [@Quency-D](https://github.com/Quency-D), I have seen that the default sampling interval applied in the algorithm is 1s. but that there could be a sampling interval of 1s and 10s (tested by Sensirion). Probably longer intervals are possible?

Thanks for the suggestion, that looks feasible, I'll look into it.

closed 2026-10-06 by jp-bennett