meshtastic/firmware #8112: [Feature Request]: Have client nodes ignore rebroadcasts by router/repeaters in their decision to rebroadcast or not

NomDeTom, 2025-09-26:
Currently even well-placed `router` nodes represent a risk to local mesh working, because they pre-empt all other retransmissions to yeet the packet as far as possible, and don't take account of potentially hidden or over-shadowed nodes that rely on a `client` rebroadcast to receive their packets.

Interim solutions such as `client_base` solve this on a pre-planned basis, and `router_late` reduces the impact of overshadowing on the first rebroadcast. However, the role of a `router` is to give a guaranteed broadcast the packet further than a normal `client` can reach. If `client` nodes hear a `router` rebroadcast, they shouldn't take it into account in their own managed-flood calculation.

Essentially, this reduces the influence (and perhaps some benefit, since we're eating into clear ChUtil for the `router`) of `router` nodes, and encourages local meshing activity.

[image]

ndoo, 2025-09-28:
Hm, the problem described would affect anyone who has a rooftop/balcony CLIENT then an indoor CLIENT right?

Looks like I should switch hilltop ROUTERs in my area to ROUTER_LATE then, since the scenario I described above is very common here.

korbinianbauer, 2025-10-03:
> node B would realize that C cannot see Router or node A

How?

Ryu945, 2025-10-05:
> > node B would realize that C cannot see Router or node A
> 
> How?

One possible implementation is for nodes to broadcast who their neighbors are similar to how telemetry is done now.

korbinianbauer, 2025-10-22:
> However, being able to quickly “shut off” nodes is unique to `ROUTER` and is very useful in certain cases (I think [@b8b8](https://github.com/b8b8) agrees). It can eliminate rebroadcasts of `CLIENT`s that contribute very little and would be covered by the `ROUTER` anyway.

I feel like as long as there's no additional cost for sending 7-hop broadcast packets vs. being nice about available airtime, it's kind of nonsensical micromanaging to try to reduce congestion by shutting down rebroadcasts unpredictably from far away. 

> However, this is all about the trade-off between unnecessary rebroadcasts and refraining from rebroadcasts too much. “Managed flooding” is a spectrum and I believe we should be careful with shifting more towards “pure flooding”.

Definitely. It would be much less painful to manage the congestion issue before even sending though, rather than having to decide when to not rebroadcast a packet that's already out and expected to arrive according to the hop_start.

I know there's some throttling already scattered over the firmware code, but it doesn't really consider channel utilization until it's too late