meshtastic/firmware #8114: [Feature Request]: Have each retransmission (first hop attempt) increase the coding rate of the radio

NomDeTom, 2025-09-26:
Similar to #8108, adding to the CR of each retransmission attempt would improve reliability of first hops.

This would be along the lines of:

Attempt 1: 4/5
Attempt 2: 4/7
Attempt 3: 4/8

An additional idea would be to increase the attempts to 4, to allow all 4 settings to be used :)

mikecarper, 2025-09-26:
4/5 and 4/7 is all you need

thebentern, 2025-09-27:
TX power is typically already set to the maximum the radio and region will allow, in most configurations. Increasing the coding rate on subsequent retransmissions would be only for "retries" when we did not receive an ACK (hear a rebroadcast). This becomes particularly useful when you have devices like the tracker cards which are quite sensitive on the RX side of things but fail to have the reach of nodes transmitting with external antennas.

korbinianbauer, 2025-09-28:
>TX power is typically already set to the maximum the radio and region will allow, in most configurations.

Yes, I'm suggesting to transmit with reduced power first to reduce interference with other nodes. Cellular does this and calls it Transmit Power Control.

This could also be combined with the idea of ramping up the CR, e.g.

First transmission: -10 dB and CR 4/5
Second transmission: -5 dB and CR 4/7
Third transmission: 0 dB (full configured tx power) and CR 4/8

I think we should be really careful with suggestions that are based on just using more airtime (e.g. higher CR or additional data, packets or retransmissions) without balacing this change by getting more conservative somewhere else whenever possible.

korbinianbauer, 2025-09-29:
>you are more likely to exhaust the hop limit sooner if the first transmission is rebroadcast by a closer client in witness of the packet vs an advantageous Router on the edge of the viewshed.

That's true for starting with a higher CR too, though.

korbinianbauer, 2025-09-29:
To be clear 4/7 is called a **lower** coding rate than 4/5, because it transmits less payload bits per transmitted bit.
A **lower** coding rate (e.g. 4/7) means more redundancy and airtime, but also less susceptibility to noise -> **more** range.

Anything that reduces range will inherently increase the chance that a closer client rebroadcasts it instead of an advantageous Router on the edge of the viewshed, because it might not hear you directly.

Any setting that increases the range of your transmission will potentially save hops, because the "advantageous Router" will hear it directly.

But it will also increase interference for nodes nearby (because it's signal is either louder or longer) and it will increase the load on the mesh as a whole (because more nodes in other directions will hear you directly and rebroadcast).

That's why I'm suggesting that if we allow more interference on the third try (by using a higher coderate), we might want to balance this out with reduced interference on the first try (lower TX power).

There's no free lunch.

If there was, there would be no reason to not just always use the lowest coderate (highest range).

mikecarper, 2025-09-29:
In my context higher meant 8 vs 5. Sorry for the confusion

mikecarper, 2025-10-23:
Dynamic power levels do not make a lot of sense in a mesh. 1:1 links it totally makes sense

NomDeTom, 2026-01-02:
Some further ideas on this:
New suggested presets are set up with CR8 as default. These would be locked out of this feature. Perhaps they need to be brought back to 6/8?

Further, the defaults should be compile-time on a per-device setting, e.g.
ON for the T1000000e, the Heltec Meshpocket, WisMesh Tag, Elecrow Thinknode M3 & M4.
OFF for any 30db or more radio
TBC for any other 22db node, potentially off by default.