meshtastic/firmware #8147: [Bug]: "Encrypted send failed" when sending direct message

Nikotine1, 2025-09-28:
### Checklist

- [x] I am able to reproduce the bug with the latest version.

- [x] I made sure that there are no existing **OPEN or CLOSED issues** which I could contribute my information to.

- [x] I have taken the time to fill in all the required details. I understand that the bug report will be dismissed otherwise.

- [x] This issue contains only one bug.

- [x] I have read and understood the **Contribution Guidelines**.

- [x] I agree to follow this project's Code of Conduct

### Affected app version

2.7.1 (29314197)

### Affected Android version

Android 15

### Affected phone model

Samsung S23 ultra

### Affected node model

Home-made Rpi shield with SX1262 modules

### Affected node firmware version

Portduino 2.6.11.60ec05e

### Steps to reproduce the bug

1. Select this one contact in my contact list
2. Send a direct message
3. Icon shows crossed-out cloud

### Actual behavior

Click cloud, message reads: Encrypted Send Failed

This is the only contact this happens with, what is the reason?

### Expected behavior

The message is sent and ideally delivered.

### Screenshots/Screen recordings

[image]

### Relevant log output

```shell
from: [redacted] (![redacted])
to: [redacted] (![redacted])
id: 3180081625
rx_time: 1759058491
priority: ACK
decoded {
  portnum: ROUTING_APP
  payload: "\030\""
  request_id: 94761420
}

Decoded Payload:
{
error_reason: PKI_FAILED

}

to: [redacted] (![redacted])
id: 94761420
want_ack: true
pki_encrypted: true
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "Hey"
}

Decoded Payload:
{
Hey
}
```

Nikotine1, 2025-09-28:
> Do you know what firmware that node is on?
> 
> Have you tried the other way of messaging in that issue ?

Unfortunately not, it's not mine.
All I can see it's a Heltec v3, but the firmware section stays empty.

GUVWAF, 2025-09-29:
> At least that is what I have seen in some limited testing, can we verify / ensure that this is not being returned if there is no matching node in the node db and send out the message?

Indeed it will return this when it is rolled out of the NodeDB. It would be a bit misleading if you still have a green lock, then send a (potentially sensitive) message, which will use the channel key, and only afterwards it turns into an open lock, no? 
Instead of the error we could potentially let it ask the client to import the contact?
Ref.:
https://github.com/meshtastic/firmware/blob/777e11bad97777d6351c04cf0e09c4983b0faff0/src/mesh/Router.cpp#L624-L625

Pedrom-pt, 2025-09-29:
I've the same issue, tested with two differents nodes.

The nodes i want to send an direct message are on my node db.
NO public key mismatch, direct message (RF direct, 0 hop), the message is not delivery "Encrypted send failed" 
My Android version: 2.7.1 (google)
My node's firmware 2.6.4

Remote node (1): Firmware 2.6.4
Remote node (2): idk

What i tried: "exchange user info and "Request Metadata"

jp-bennett, 2025-09-29:
I think we can actually handle this case by re-adding the contact to the nodeDB from the information provided from the client, and then immediately use that to send the DM.

Stephan-2025, 2025-10-01:
I´ve that issue too since update on Android to 2.7.1 APP Version on a RAK Device (Firmware 2.6.6), which worked fine bevor the update of the App. Also deleting the contact out of my node db don´t solve it. But the encryption-sign is green ... but i´m able to send messages out in channel - no problem there. I also tried another node - the same PN Problem.
But only PNs are no more possible to send, on channels everythink works fine

HIlight3R, 2025-12-14:
Android client, same issue on T-Echo 2.7.16

otakup0pe, 2026-01-30:
I am able to consistently send DM's to a low-hop node with no issue, however at higher hops I get `Encrypted Send Failed`. This seems to rule out public key issues. Happy to do more troubleshooting, if there are suggested next steps.

domusonline, 2026-04-02:
This can be reproduced as mentioned above by removing the node from the NodeDB on the radio (through CLI for example). We would need to:
1- Understand what would be the solution for those cases (it was discussed that the node data could be send from the application and added automatically). I believe this is the most common cause because a simple remove and add of the node again usually works

2- Understand if there are other cases where the message is shown and there is no difference in the key

erayd, 2026-04-03:
> It is an intended behavior for a security feature with a solution, the client (which is the source of the the noddeid that is not in the device node database) needs to send a contact to the device before it sends every dm. The resolution to this requires firmware that supports adding contacts and a recent version of the android or iOS app. If you are seeing this issue with other clients an issue needs to be opened in the appropriate place.

@garthvh Can you please stop prematurely closing this issue, and reopen it again. It's clearly a problem, and multiple people are indicating that they can reproduce it. It's great that there's an intended behaviour that is supposed to be allowing DMs to be sent to a node that is not in the device database (and your context about why the particular point in the code identified upthread is not where the fix needs to be made is helpful), however that behaviour is evidently *not reliably working*, and is causing real-world problems for users. Which is therefore still obviously a bug that needs fixing.

Good intentions do not magically make a broken feature not broken, and all closing the issue (especially as completed) does is try to sweep the whole thing under the carpet instead of actually solving it. Which I note that others in this thread were actually trying to do until you jumped in and closed the thing. It should remain open until the problem is actually fixed.

I can confirm that this problem occurs frequently here with FW 2.7.15 and recent Android app versions. So have other users on this issue. I've not specifically tried to gather data more widely than that, but could probably get it if necessary.

Having now also looked at the code for `NodeDB::addFromContact`, I can't see any recent changes to it that appear relevant, nor has a PR search revealed anything that appears to be a fix. Is there something that you believe fixes this issue somewhere else in a more recent firmware version?

/CC @thebentern

garthvh, 2026-04-03:
Then open an issue in the Android repo where the fix is not working, the firmware is doing its job here.