meshtastic/firmware #8153: [Feature Request]: Introduce a metric of some kind to indicate when a mesh may be reaching a threshold for moving to a faster preset

NomDeTom, 2025-09-28:
A node could gauge the airtime that itself and adjacent nodes are using, or the channel utilisation, or generally the time they spend throttling messages to work out whether there is high airtime contention generally. With this figure, and some kind of link-strength metric, it would be possible to see if a faster preset would be possible, desirable or useful.

Ryu945, 2025-10-02:
Devices generally have to reboot between Lora settings so swapping back and forth in a short amount of time is not viable.  Also, mesh networks can be huge and there could one part of the network where you can swap to something faster but another part where you cannot.  This sounds like it may be more impractical then it is worth the time working on because there would never be any consensus on the best Lora setting.  Thus, you often have the slower and longer range setting as everyone's default (Long Fast).

It seems like optimizing the bandwidth might be a better use of time though there is a mode I think would be useful for this concept because it doesn't require the network to change.

Some Meshtastic devices can store messages and send them later.  I know ESP32 can.  I could see someone setting their device up to store messages when on one channel setting for some period of time.  Then rebooting with a different Lora configuration and broadcasting all the messages under that.  Then storing messages for awhile on that Lora setup.  It would constantly swap back and forth rebooting.

Practically though, you might be better off just buying two devices and wire them together (Or one device that can operate both at once.  This devices does exist).  Then set them up to relay messages between each other.  It will probably take some new software developed.  I have thought about this idea with the Chirp chips that work both on Lora frequencies and Wifi frequencies.  Wifi has the bandwidth but terrible range.  Lora frequencies has the range but terrible bandwidth.  I have suspected Wifi frequencies set to max range will offer a better trade off then Lora frequencies set to short range and higher bandwidth.  

If I remember correctly, a desert environment with wifi frequencies set to Long Slow (though I think there is a better way to optimize Wifi since the channels are wider) has the same range as somewhere between a Lora frequencies set to Medium Slow and Medium Fast.  If your on a network that is trying to use Short Fast or Short Slow, would wifi frequencies be better anyway?

korbinianbauer, 2025-10-03:
> Mesh networks can be huge and there could one part of the network where you can swap to something faster but another part where you cannot. This sounds like it may be more impractical then it is [...]

Yes, but in practice some regions DID switch to faster preset and are happy with the result.

The nodes suggesting it would be advantageous to make the move should make the collective effort more likely to succeed.

NomDeTom, 2025-10-27:
Some of the functionality for this is contained in #8455

NomDeTom, 2025-12-12:
- Average number of retries

- Average number of acks to a flood packet..

alecperkins, 2026-03-11:
(Sharing a perspective from [the middle](https://nyme.sh/preset-testing/) of such a transition.)

There are two angles for this question:
1. Is the network ready for a bump?
2. Is _this_ node proximate enough to density for a bump to work for its links?

A lot of the awareness of needing to upgrade comes from a higher level observation of the network that I don’t think a single node can figure out conclusively. But:

Sustained high channel utilization definitely supports the first question. If a node can go more than a day without ever broadcasting its telemetry due to Channel Utilization-throttling then the network is definitely ready for an upgrade. _Especially_ if it’s a router-type. That was one of the big metrics local operators were using, their solar ROUTER nodes were never sending telemetry because they were jammed up with packets. I think we were averaging about 40% utilization continuously.

ROUTER_LATEs also tended to be unusable because their TX queue would fill up too quickly, to the point we could not remote admin them.

Comparing the utilization against the number of text packets was also something we used to gauge the need to move, though that is dependent on both the purpose of the network and also the interval configs. Also it’s mostly relevant for the _initial_ move, less so the subsequent one, as we are being strict about non-text packet intervals. Perhaps this belongs in a dedicated module that can be opted-into, NetworkStats or something.

Another metric that was a good indicator was the last-heard of nodes in the nodedb. We got to a point where we could reset our nodedb and it would be refilled within the hour (including unknowns). This of course won’t change after the first migration assuming most people move, but it’s a good indicator of needing to get off the current defaults. Maybe it could be calibrated to the channel utilization or typical airtime so it could continue to be a useful indicator for successive preset changes, or scaled against the default: SF / 11 x 60 min.

For the second question, the presets mostly change the importance of RSSI rather than SNR (SNR will generally go _up_ with shorter SF since the noise denominator is lower). What will matter is their link budget will go down so links that are too weak will fall off. Maybe the heuristic includes if there are at least two direct nodes with a consistent inbound RSSI high enough that a faster setting would still come through? Specifically if the RSSI plus the processing gain at the lower spread factor will keep the link above the noise floor. I don’t know if most nodes have sufficient resources to track and compute that.

The sheer number of directs also contributes to the second question, combined with the RSSI it should give a reasonable picture of density. Perhaps with some further check on the number of each role it sees direct, and its own role.

And ultimately, where does this show up? If a remote node detects this condition does it advertise some admin packet which the admin nodes detect and surface in the apps? A non-admin `HELP_IM_DROWNING` portnum? That could be useful for tools like Meshview or Malla which could watch for those from multiple nodes and surface an alert. Some of this analysis could probably be handled by the apps rather than the nodes themselves.

Some additional features that maybe would support this:
- Top-level config that chooses between infrastructure and ad hoc usecases, so the node defaults can switch based on the kind of network it’s intended to be used with. This is probably more of an app/web UI question.
- Temporary configs that automatically revert, so it’s easier to test radio configs without risk of the remote node becoming unreachable. Make it easy for people to confirm that a preset change is viable. This was probably the biggest hurdle for certain operators. We probably could have gone straight to MediumFast even but were reluctant to try given the uncertainty about recovering hard-to-access nodes if it didn’t work.